kmod-project/kmod #3: depmod updates modules.* files even when not needed

evelikov, 2021-08-04:
Seems like depmod will eagerly overwrite many of the existing modules.* files, even if there are no changes vs the existing files.
Can we fix the tool to write the files, only as needed?

Thanks in advance

Simple reproducer:
```bash
#!/bin/bash

echo "1. Fetch checksums - file timestamps and contents"
mkdir before
for i in /usr/lib/modules/$(uname -r)/modules.*; do
  mname=$(basename $i)
  ls -la $i | sha256sum > before/$mname-ls.sum
  sha256sum $i > before/$mname.sum
done

echo "2. Kick off depmod"
depmod $(uname -r)

echo "3. Re-fetch checksums - file timestamps and contents"
mkdir after
for i in /usr/lib/modules/$(uname -r)/modules.*; do
  mname=$(basename $i)
  ls -la $i | sha256sum > after/$mname-ls.sum
  sha256sum $i > after/$mname.sum
done

echo "4. Compare - observe how most module.* timestamps have changed, yet contents are identical"
$ diff -rq before after
```
The changes ->
Files before/modules.alias.bin-ls.sum and after/modules.alias.bin-ls.sum differ
Files before/modules.alias-ls.sum and after/modules.alias-ls.sum differ
Files before/modules.builtin.alias.bin-ls.sum and after/modules.builtin.alias.bin-ls.sum differ
Files before/modules.builtin.bin-ls.sum and after/modules.builtin.bin-ls.sum differ
Files before/modules.dep.bin-ls.sum and after/modules.dep.bin-ls.sum differ
Files before/modules.dep-ls.sum and after/modules.dep-ls.sum differ
Files before/modules.devname-ls.sum and after/modules.devname-ls.sum differ
Files before/modules.softdep-ls.sum and after/modules.softdep-ls.sum differ
Files before/modules.symbols.bin-ls.sum and after/modules.symbols.bin-ls.sum differ
Files before/modules.symbols-ls.sum and after/modules.symbols-ls.sum differ

lucasdemarchi, 2022-06-30:
I'm a little bit confused that you say depmod updates it even if it's not needed. That is statement is true: we write to a temporary file and if successful we rename it to the final index name without comparing to what is currently there.

But in your case

> The changes ->
> Files before/modules.alias.bin-ls.sum and after/modules.alias.bin-ls.sum differ
> Files before/modules.alias-ls.sum and after/modules.alias-ls.sum differ
> Files [...]

it was actually different... ?

evelikov, 2022-07-04:
Note that the `foo-ls.sum` files contain the filename + timestamps. Aka the output of `ls -la foo`

In hindsight using `foo-ls` files might have been a bad choice - sorry about that.

evelikov, 2023-03-07:
In my view, kmod should not be overwriting the files if they are identical. Amongst others this confuses tooling which check the timestamps (aka pacman -Qkk), plus the files might be on a RO partition in the first place.

Admittedly the latter is a bit of a moot point and better addressed with the outdir option aka https://github.com/kmod-project/kmod/issues/13. And we got around the former by not shipping the respective files in the package.

There is value in the proposal, although my immediate "pains" are dealt with so I'm going to close this. Feel free to reopen and/or pursue.

closed 2023-03-07 by evelikov