kmod-project/kmod #4: Warning with test-depmod

rprasanth, 2021-08-17:
Hi Team,

I am trying to run the test suite for test-depmod,
I could see the Warning for all except 'depmod_modules_order_for_compressed'  I could see modules.order and modules.builtin were available in the rootfs for 'depmod_modules_order_for_compressed' alone.
https://github.com/kmod-project/kmod/tree/master/testsuite/rootfs-pristine/test-depmod/modules-order-compressed/lib/modules/4.4.4

```
# ./testsuite/test-depmod
TESTSUITE: running depmod_search_order_override, in forked context
depmod: WARNING: could not open modules.order at /lib/modules/4.4.4: No such file or directory
depmod: WARNING: could not open modules.builtin at /lib/modules/4.4.4: No such file or directory
TESTSUITE: 'depmod_search_order_override' [2746] exited with return code 0
TESTSUITE: PASSED: depmod_search_order_override
TESTSUITE: ------
TESTSUITE: running depmod_search_order_external_last, in forked context
depmod: WARNING: could not open modules.order at /lib/modules/4.4.4: No such file or directory
depmod: WARNING: could not open modules.builtin at /lib/modules/4.4.4: No such file or directory
TESTSUITE: 'depmod_search_order_external_last' [2747] exited with return code 0
TESTSUITE: PASSED: depmod_search_order_external_last
TESTSUITE: ------
TESTSUITE: running depmod_search_order_external_first, in forked context
depmod: WARNING: could not open modules.order at /lib/modules/4.4.4: No such file or directory
depmod: WARNING: could not open modules.builtin at /lib/modules/4.4.4: No such file or directory
TESTSUITE: 'depmod_search_order_external_first' [2748] exited with return code 0
TESTSUITE: PASSED: depmod_search_order_external_first
TESTSUITE: ------
TESTSUITE: running depmod_detect_loop, in forked context
TESTSUITE: ERR: 'depmod_detect_loop' [2749] exited with return code 1
TESTSUITE: EXPECTED FAIL: depmod_detect_loop
TESTSUITE: ------
TESTSUITE: running depmod_search_order_same_prefix, in forked context
depmod: WARNING: could not open modules.order at /lib/modules/4.4.4: No such file or directory
depmod: WARNING: could not open modules.builtin at /lib/modules/4.4.4: No such file or directory
TESTSUITE: 'depmod_search_order_same_prefix' [2750] exited with return code 0
TESTSUITE: PASSED: depmod_search_order_same_prefix
TESTSUITE: ------
TESTSUITE: running depmod_search_order_simple, in forked context
depmod: WARNING: could not open modules.order at /lib/modules/4.4.4: No such file or directory
depmod: WARNING: could not open modules.builtin at /lib/modules/4.4.4: No such file or directory
TESTSUITE: 'depmod_search_order_simple' [2751] exited with return code 0
TESTSUITE: PASSED: depmod_search_order_simple
TESTSUITE: ------
TESTSUITE: running depmod_modules_order_for_compressed, in forked context
TESTSUITE: 'depmod_modules_order_for_compressed' [2752] exited with return code 0
TESTSUITE: PASSED: depmod_modules_order_for_compressed
TESTSUITE: ------
# 
```

May I know, how to avoid the warning?

lucasdemarchi, 2022-06-22:
Don't run as root. If you are running the tests manually, you need to re-generate the rootfs by calling `make rootfs`, whcih takes a few seconds only.

lucasdemarchi, 2022-06-30:
Closing. Let me know if the issue persists and we can re-open it. Thanks.

closed 2022-06-30 by lucasdemarchi