kmod-project/kmod #10: lsmod: fallback to procfs

jengelh, 2022-01-04:
On Linux 5.15.12, kmod-29, inside an nspawn container, I observe, somewhat expectedly:

```
$ lsmod 2>&1 | grep twofish_comm
libkmod: kmod_module_get_holders: could not open '/sys/module/twofish_common/holders': No such file or directory
twofish_common            -2  -2
```

Since `/proc/modules` is already opened by libkmod to discover the list of loaded modules, wouldn't it be possible to also source the other information from there as a fallback?

```
$ grep twofish_com /proc/modules
twofish_common 24576 4 twofish_generic,twofish_avx_x86_64,twofish_x86_64_3way,twofish_x86_64, Live 0x0000000000000000 (E)
```

lucasdemarchi, 2022-06-30:
Cc'ing @poettering

jengelh, 2022-06-30:
Quoting from https://github.com/systemd/systemd/issues/21744#issuecomment-991849471 : “Sysfs is simy not virtualized for cotnainers”, which is probably why containers have a `/sys` path that is a tmpfs with only a few select directories in it which are bind-mounted… but `/sys/module` is [not one of them](https://github.com/systemd/systemd/blob/master/src/nspawn/nspawn-mount.c#L434).

poettering, 2022-06-30:
Yes, precisely, we mount /sys/ as tmpfs into the container, and then some select subtrees into it, but not /sys/modules/. That was a concious decision really...

I wonder if we should hide /proc/modules in nspawn containers too btw...

lucasdemarchi, 2022-06-30:
@poettering but what I said is: if you want to have visibility on loaded modules from inside the container you could mount it as read-only. That does not contradict what is written at http://systemd.io/CONTAINER_INTERFACE/ :

"Make sure to pre-mount /proc/, /sys/, and /sys/fs/selinux/ before invoking systemd, and mount /sys/, /sys/fs/selinux/ and /proc/sys/ read-only (the latter via e.g. a read-only bind mount on itself)"

lucasdemarchi, 2022-06-30:
The *default* behavior in systemd-nspawn to hide both would make sense indeed. And users who want it should be able to mount sysfs as read-only

closed 2025-02-25 by lucasdemarchi