kmod-project/kmod #13: depmod: Allow output file location to be overridden

AdrianVovk, 2022-06-20:
I've got a package build system that looks something like this:
- stage dependencies into /
- make / read-only (this is necessary not just to make sure we're not losing build output, but also due to the way build dependencies are stored in CAS)
- stage sources into /work/dir, mark it writable. Create writeable /installdir
- run build commands
- everything inside of /installdir is the "output" of this package. Everything else gets discarded

I'd like to make a package that depends on the kernel & all the external modules I want, and runs depmod for this configuration. Then I can glue everything together at a later point (out of scope for this issue)

`depmod` isn't making this easy, because it always writes its output files into the kernel's directory. Since the kernel & modules are staged at `/usr/lib/modules`, and `/usr/lib/modules` is (necessarily) read-only, `depmod` will try and fail to write things into `/usr/lib/modules/VERSION`. I'd like a way to tell depmod to actually write into `/installdir/usr/lib/modules/VERSION`, or somewhere inside of `/work/dir` so that I could then manually `install` the files into the right place. 

For now, we've got a workaround that looks something like this: 
- link `/usr/lib/modules/VERSION/kernel` into `/installdir/usr/lib/modules/VERSION/kernel`
- Copy `/usr/lib/modules/VERSION/modules.order` into `/installdir/usr/lib/modules/VERSION/modules.order`
- Same with `modules.builtin`
- Run `depmod -a -b /installdir/usr VERSION`
- Delete `/installdir/usr/lib/modules/VERSION/{kernel,modules.order,modules.builtin}`
- The files created by `depmod` remain in `/installdir/usr/lib/modules/VERSION`, as we'd like

Potential solutions for `depmod`:
- depmod could respect an environment variable like `DESTDIR` or similar (like in autotools: for example in my use-case above `DESTDIR=/installdir`)
- depmod could add a command line flag that plays a role similar to the `DESTDIR` env var
- depmod could add a command line flag to just dump the files into the current working directory. From there they could manually be installed into wherever they need to go

This is a "golden image" use-case. There is no way `depmod` could run from userspace like it does in most other distros

evelikov, 2023-01-31:
Opened a PR as well, so the maintainers go with either option.

evelikov, 2023-02-09:
Landed with https://github.com/kmod-project/kmod/commit/1712a1548eeaad61143f303b09afcb4215943203. This issue can be closed.

closed 2023-02-09 by lucasdemarchi