kmod-project/kmod #16: testsuite failures on mipsel

mbiebl, 2022-07-19:
The latest upload of kmod in Debian enabled the testsuite to be run during build. This caused the testsuite to fail on mipsel.
The full build log is available at https://buildd.debian.org/status/fetch.php?pkg=kmod&arch=mipsel&ver=30%2B20220630-2&stamp=1657943365&raw=0

```
===============================
   kmod 30: ./test-suite.log
===============================

# TOTAL: 16
# PASS:  12
# SKIP:  0
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: testsuite/test-modinfo
============================

modinfo: ERROR: Module /mod-simple-sha1.ko not found.
modinfo: ERROR: Module /mod-simple-sha256.ko not found.
modinfo: ERROR: Module /mod-simple-pkcs7.ko not found.
TESTSUITE: running test_modinfo_external, in forked context
TESTSUITE: 'test_modinfo_external' [1230957] exited with return code 0
TESTSUITE: PASSED: test_modinfo_external
TESTSUITE: ------
TESTSUITE: running test_modinfo_sig_hashalgo, in forked context
TESTSUITE: ERR: 'test_modinfo_sig_hashalgo' [1230970] exited with return code 1
TESTSUITE: ERR: FAILED: test_modinfo_sig_hashalgo
TESTSUITE: ------
FAIL testsuite/test-modinfo (exit status: 1)

FAIL: testsuite/test-modprobe
=============================

libkmod: ERROR ../libkmod/libkmod-config.c:489 kcmdline_parse_result: Ignoring bad option on kernel command line while parsing module name: 'ivrs_acpihid[14:00'
modprobe: ../testsuite/init_module.c:160: create_sysfs_files: Assertion `mkdir_p(buf, len, 0755) >= 0' failed.
TESTSUITE: running modprobe_external, in forked context
TESTSUITE: 'modprobe_external' [1230997] exited with return code 0
TESTSUITE: PASSED: modprobe_external
TESTSUITE: ------
TESTSUITE: running modprobe_oldkernel_force, in forked context
TESTSUITE: 'modprobe_oldkernel_force' [1231004] exited with return code 0
TESTSUITE: PASSED: modprobe_oldkernel_force
TESTSUITE: ------
TESTSUITE: running modprobe_oldkernel, in forked context
TESTSUITE: 'modprobe_oldkernel' [1231009] exited with return code 0
TESTSUITE: PASSED: modprobe_oldkernel
TESTSUITE: ------
TESTSUITE: running modprobe_force, in forked context
TESTSUITE: 'modprobe_force' [1231014] exited with return code 0
TESTSUITE: PASSED: modprobe_force
TESTSUITE: ------
TESTSUITE: running modprobe_param_kcmdline8, in forked context
TESTSUITE: 'modprobe_param_kcmdline8' [1231018] exited with return code 0
TESTSUITE: PASSED: modprobe_param_kcmdline8
TESTSUITE: ------
TESTSUITE: running modprobe_param_kcmdline7, in forked context
TESTSUITE: 'modprobe_param_kcmdline7' [1231027] exited with return code 0
TESTSUITE: PASSED: modprobe_param_kcmdline7
TESTSUITE: ------
TESTSUITE: running modprobe_param_kcmdline6, in forked context
TESTSUITE: 'modprobe_param_kcmdline6' [1231028] exited with return code 0
TESTSUITE: PASSED: modprobe_param_kcmdline6
TESTSUITE: ------
TESTSUITE: running modprobe_param_kcmdline5, in forked context
TESTSUITE: 'modprobe_param_kcmdline5' [1231029] exited with return code 0
TESTSUITE: PASSED: modprobe_param_kcmdline5
TESTSUITE: ------
TESTSUITE: running modprobe_param_kcmdline4, in forked context
TESTSUITE: 'modprobe_param_kcmdline4' [1231030] exited with return code 0
TESTSUITE: PASSED: modprobe_param_kcmdline4
TESTSUITE: ------
TESTSUITE: running modprobe_param_kcmdline3, in forked context
TESTSUITE: 'modprobe_param_kcmdline3' [1231031] exited with return code 0
TESTSUITE: PASSED: modprobe_param_kcmdline3
TESTSUITE: ------
TESTSUITE: running modprobe_param_kcmdline2, in forked context
TESTSUITE: 'modprobe_param_kcmdline2' [1231032] exited with return code 0
TESTSUITE: PASSED: modprobe_param_kcmdline2
TESTSUITE: ------
TESTSUITE: running modprobe_param_kcmdline_show_deps, in forked context
TESTSUITE: 'modprobe_param_kcmdline_show_deps' [1231033] exited with return code 0
TESTSUITE: PASSED: modprobe_param_kcmdline_show_deps
TESTSUITE: ------
TESTSUITE: running modprobe_install_cmd_loop, in forked context
TESTSUITE: ERR: 'modprobe_install_cmd_loop' [1231034] terminated by signal 6 (Aborted)
TESTSUITE: ------
FAIL testsuite/test-modprobe (exit status: 1)

FAIL: testsuite/test-blacklist
==============================

TESTSUITE: running blacklist_1, in forked context
TESTSUITE: ERR: 'blacklist_1' [1231005] exited with return code 1
TESTSUITE: ERR: FAILED: blacklist_1
TESTSUITE: ------
FAIL testsuite/test-blacklist (exit status: 1)

FAIL: testsuite/test-depmod
===========================

depmod: WARNING: could not open modules.order at /lib/modules/4.4.4: No such file or directory
depmod: WARNING: could not open modules.builtin at /lib/modules/4.4.4: No such file or directory
depmod: WARNING: could not open modules.builtin.modinfo at /lib/modules/4.4.4: No such file or directory
depmod: WARNING: could not open modules.order at /lib/modules/4.4.4: No such file or directory
depmod: WARNING: could not open modules.builtin at /lib/modules/4.4.4: No such file or directory
depmod: WARNING: could not open modules.builtin.modinfo at /lib/modules/4.4.4: No such file or directory
TESTSUITE: running depmod_search_order_override, in forked context
TESTSUITE: 'depmod_search_order_override' [1231007] exited with return code 0
TESTSUITE: PASSED: depmod_search_order_override
TESTSUITE: ------
TESTSUITE: running depmod_search_order_external_last, in forked context
TESTSUITE: 'depmod_search_order_external_last' [1231010] exited with return code 0
TESTSUITE: ERR: sizes do not match /<<PKGBUILDDIR>>/build-deb/testsuite/rootfs/test-depmod/search-order-external-last/lib/modules/4.4.4/correct-modules.dep /<<PKGBUILDDIR>>/build-deb/testsuite/rootfs/test-depmod/search-order-external-last/lib/modules/4.4.4/modules.dep
TESTSUITE: ERR: FAILED: exit ok but outputs do not match: depmod_search_order_external_last
TESTSUITE: ------
FAIL testsuite/test-depmod (exit status: 1)

============================================================================
Testsuite summary for kmod 30
============================================================================
# TOTAL: 16
# PASS:  12
# SKIP:  0
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
Please report to linux-modules@vger.kernel.org
============================================================================
```

mbiebl, 2022-07-19:
https://wiki.debian.org/ArchitectureSpecificsMemo

mbiebl, 2022-07-19:
contents of builddir/testsuite/
<details>

```
testsuite/
testsuite/stamp-rootfs
testsuite/test-hash.trs
testsuite/test-depmod.trs
testsuite/.dirstamp
testsuite/test-new-module.log
testsuite/delete_module.lo
testsuite/.deps
testsuite/.deps/.dirstamp
testsuite/test_new_module-test-new-module.o
testsuite/test-hash.log
testsuite/libtestsuite_la-testsuite.lo
testsuite/test-testsuite.trs
testsuite/init_module.la
testsuite/test-initstate
testsuite/test-loaded.trs
testsuite/test-initstate.log
testsuite/test-testsuite
testsuite/test-util.trs
testsuite/test-array.log
testsuite/test-initstate.trs
testsuite/test_modprobe-test-modprobe.o
testsuite/.libs
testsuite/.libs/path_la-path.o
testsuite/.libs/init_module.so.0.0.0
testsuite/.libs/delete_module.so.0.0.0
testsuite/.libs/init_module.so.0
testsuite/.libs/path.so.0.0.0
testsuite/.libs/uname.o
testsuite/.libs/delete_module.so
testsuite/.libs/init_module.la
testsuite/.libs/libtestsuite_la-testsuite.o
testsuite/.libs/delete_module.la
testsuite/.libs/uname.so
testsuite/.libs/uname.la
testsuite/.libs/delete_module.so.0
testsuite/.libs/path.lai
testsuite/.libs/uname.lai
testsuite/.libs/delete_module.lai
testsuite/.libs/init_module.o
testsuite/.libs/path.la
testsuite/.libs/delete_module.o
testsuite/.libs/init_module.lai
testsuite/.libs/uname.so.0
testsuite/.libs/libtestsuite.a
testsuite/.libs/libtestsuite.la
testsuite/.libs/init_module.so
testsuite/.libs/path.so
testsuite/.libs/path.so.0
testsuite/.libs/uname.so.0.0.0
testsuite/delete_module.la
testsuite/uname.la
testsuite/test-dependencies.trs
testsuite/test-strbuf.trs
testsuite/test_hash-test-hash.o
testsuite/uname.lo
testsuite/test-util.log
testsuite/test-depmod
testsuite/test-loaded
testsuite/test-modprobe
testsuite/test-modprobe.log
testsuite/test-hash
testsuite/path_la-path.lo
testsuite/test-dependencies.log
testsuite/test_depmod-test-depmod.o
testsuite/test-blacklist
[... 672 lines cut ...]
testsuite/rootfs/test-init/mod-simple.ko
testsuite/rootfs/test-init/sys
testsuite/rootfs/test-init/sys/module
testsuite/rootfs/test-init/sys/module/mod_simple
testsuite/rootfs/test-init/sys/module/mod_simple/initstate
testsuite/init_module.lo
testsuite/test_testsuite-test-testsuite.o
testsuite/test-strbuf.log
testsuite/test-testsuite.log
testsuite/test_strbuf-test-strbuf.o
testsuite/test-modinfo.trs
testsuite/test-strbuf
testsuite/test_loaded-test-loaded.o
testsuite/test-list.trs
testsuite/test-array
testsuite/test-list
testsuite/test-blacklist.log
testsuite/test_init-test-init.o
testsuite/test-init
testsuite/test-depmod.log
testsuite/test_blacklist-test-blacklist.o
testsuite/test-modinfo.log
```

</details>

lucasdemarchi, 2022-07-19:
Humn... and in theory we have a pass on the test of the testuite, `PASS: testsuite/test-testsuite`, which verifies the chroot is working properly.

lucasdemarchi, 2022-07-19:
could you tar the rootfs and make it available somewhere for me to try it manually?

mbiebl, 2022-07-19:
In case you have a Debian environment:
https://wiki.debian.org/EmDebian/CrossDebootstrap#QEMU.2Fdebootstrap_approach

This should allow you to create a mipsel chroot enviroment running via qemu.

lucasdemarchi, 2022-07-19:
@mbiebl no need to upload it. Thanks for the detailed instructions, I will try it soon (probably tomorrow).

mbiebl, 2022-07-23:
If there's anything I can do to help you with this issue, please let me know.

lucasdemarchi, 2022-08-07:
Sorry for the delay. I went on a trip, returning this week.

From a few debug statements added, it seems our LD_PRELOAD'ed path.so is failing to intercept the stat() call, so is_module_filename() in modinfo.c returns false. I do get a few warnings in this build:

```
../testsuite/path.c:201:14: warning: no previous prototype for '__xstat' [-Wmissing-prototypes]
  201 | WRAP_VERSTAT(__x,);
      |              ^~~
../testsuite/path.c:166:15: note: in definition of macro 'WRAP_VERSTAT'
  166 | TS_EXPORT int prefix ## stat ## suffix (int ver,            \
      |               ^~~~~~
../testsuite/path.c:202:14: warning: no previous prototype for '__lxstat' [-Wmissing-prototypes]
  202 | WRAP_VERSTAT(__lx,);
      |              ^~~~
../testsuite/path.c:166:15: note: in definition of macro 'WRAP_VERSTAT'
  166 | TS_EXPORT int prefix ## stat ## suffix (int ver,            \
      |               ^~~~~~
```

I will investigate more.

lucasdemarchi, 2022-08-08:
@mbiebl  was there any change in the libc? The stat() behavior seems odd:

```
TESTSUITE: running test_modinfo_sig_hashalgo, in forked context
is_module_filename stat
wrap - stat(/mod-simple-sha1.ko)
wrap - stat: rootpath=/root/kmod-30+20220630/build-deb/testsuite/rootfs/test-modinfo/
wrap - stat: path=/root/kmod-30+20220630/build-deb/testsuite/rootfs/test-modinfo//mod-simple-sha1.ko
wrap - libc_stat: ret=0
is_module_filename /mod-simple-sha1.ko ret=0 file-type=00000
```

So... we are actually wrapping stat() correctly, stat() is returning 0 as it should, but file-type (st.st_mode & S_IFMT) is returning 0.

mbiebl, 2022-08-08:
> @mbiebl was there any change in the libc? The stat() behavior seems odd:

dunno. the test suite was only enabled recently. It's possible that it has been broken for a longer time

rfc1036, 2022-08-08:
I do not remember the details but at the time I had checked, and it was been broken also for VERY old versions of kmod.

For the time being I have disabled the test suite on mipsel.

jcristau, 2022-09-02:
I think what's going on is this: normally when you call `stat()` with `_FILE_OFFSET_BITS=64`, `<sys/stat.h>` redirects that to the `stat64` function.  Because we go through `get_libc_func`, that returns the actual `stat` function, which works with 32-bit `ino_t`, so there's a struct layout mismatch with what the test expects, and the st_mode field ends up being garbage.

lucasdemarchi, 2022-09-05:
@jcristau  yeah, that makes sense. On my system stat is simply a `#define stat stat64`.  For path.c and that file only, it makes sense to unconditionally override the function from libc.

commit b4d281f9 (2022-09-05): testsuite: fix override of `stat` on 32-bit architectures

When _FILE_OFFSET_BITS is 64, glibc headers turn `stat` calls into
`stat64`, and our `stat` override into a `stat64` function.  However,
because we use dlsym to get the address of libc's `stat`, we end up
calling into the "real" `stat` function, which deals with 32-bit off_t,
and we treat its result as if it were returned from stat64.  On most
architectures this seems to have been harmless, but on 32-bit mips,
st_mode's offset in struct stat and struct stat64 are different, so we
read garbage.

To fix this, explicitly unset _FILE_OFFSE [... cut]
closed 2022-09-05 by lucasdemarchi