kmod-project/kmod #17: built-in module is loaded instead of the duplicated one in "extra" dir

winndows, 2022-08-22:
Hello,

As we know, module in "extra" has higher priority than the duplicate one in kernel.

But it looks like when the readdir() read sub-dirs of /lib/modules/5.18.0-0.bpo.1-rt-amd64/, it does not always read out "extra" before "kernel" and this makes ko path in built-in kernel is updated into modules.dep, but what we want is the path in "extra".

I was using isar: https://github.com/ilbers/isar to build a system img
and in the img, i have one ko as kernel built-in and a duplicated one in "extra"
before start the img, i chroot into the rootfs of the img, but when I run "depmod -a $(ls /lib/modules/)", it will put built-in ko into modules.dep and I found readdir read kernel dir firstly instead of extra dir.

kernel is read firstly by readdir():
[image]

[image]

module in extra is set as lower prio:
[image]

lucasdemarchi, 2022-08-22:
built-in is always highest prio as it means the module is in the kernel image rather than being built as a module. You can't override that with a module. If you need that, you need to change your kernel config to make that kconfig entry a module.

lucasdemarchi, 2022-08-22:
Although from your logs it's actually using a module. Did you mix the built-in index with another kernel or is just the word used the threw me off?

lucasdemarchi, 2022-08-22:
From your logs I also see you don't have any depmod.d. A config in one of these directories is where depmod learns the search priority from so you probably need to add one such file

winndows, 2022-08-23:
> built-in is always highest prio as it means the module is in the kernel image rather than being built as a module. You can't override that with a module. If you need that, you need to change your kernel config to make that kconfig entry a module.

sorry, "built-in" I mean the module is built with kernel and installed in the /lib/modules/`uname -r`/kernel/, but it is a module, not in the vmlinuz img.  "extra" means modules are installed in the /lib/modules/`uname -r`/extra/

winndows, 2022-08-23:
> From your logs I also see you don't have any depmod.d. A config in one of these directories is where depmod learns the search priority from so you probably need to add one such file

I have another system(vmware vm using the img  I built), instead of the one I mentioned before(rootfs of isar img), when I do depmod, the readdir will always read extra firstly, then ko in extra path will updated to modules.dep.

Yes, I don't have any depmod.d for both of them. adding such a file can be a workaround, but I hope you expert could help to find the root cause.

[image]

[image]

winndows, 2022-08-23:
Currently, the situation is:

1. in chroot env of img rootfs, run depmod, readdir will read "kernel" firstly, then update ko in kernel dir to modules.dep
2. in vm or real HW, run depmod, readdir will read "extra" firstly, then update ko in extra dir to modules.dep

So I suspect "readdir" is not trustable here.

lucasdemarchi, 2022-08-23:
you can't rely on readdir order. If you don't set a search order to change the priorities of each of the directories, they will have the same priority and any of them will be picked up.

There is one exception: if there's no configuration the default is to use the order "updates built-in". If you don't want the configuration file, then I suggest using that name.

```
           By default, depmod will give a higher priority to a directory with
           the name updates using this built-in search string: "updates
           built-in" but more complex arrangements are possible and are used
           in several popular distributions.
```

This is what Manjaro has:

```
$ cat /usr/lib/depmod.d/search.conf 
#
# /usr/lib/depmod.d/search.conf
#

search updates extramodules built-in
```

And Ubuntu:

```
$ cat /etc/depmod.d/ubuntu.conf 
search updates ubuntu built-in
```

winndows, 2022-08-23:
Thanks a lot for the info.

One more concern, what does the "built-in" mean in the following?
`$ cat /etc/depmod.d/ubuntu.conf 
search updates ubuntu built-in`

I was thinking that modules whose code is with kernel are called built-in: no matter if they are built out in vmlinux or a separated moudule in /lib/modules/`uname -r`/kernel.

winndows, 2022-08-25:
Thanks a lot.

closed 2022-08-25 by winndows