kmod-project/kmod #26: Segmentation fault when running depmod/kmod

martinclauss, 2023-03-16:
Hi :)

I first reported the bug here: https://bugs.archlinux.org/task/77868 but since it is an upstream issue, I want to report it here as well.

I built a custom Linux kernel with debugging information. If I'm correct the bug occurs when keeping the debug information in the modules. Excerpt from the PKGBUILD file (Arch Linux):

```
  echo "Installing modules..."
  make INSTALL_MOD_PATH="$pkgdir/usr" INSTALL_MOD_STRIP="--only-keep-debug" \
    DEPMOD=/doesnt/exist modules_install  # Suppress depmod
```

`INSTALL_MOD_STRIP="--only-keep-debug"` was `INSTALL_MOD_STRIP=1` before the modification.

With

```
gdb -ex "set disassembly-flavor intel" -ex "run" -ex "bt full" -ex "info registers" -ex 'x/50i $pc' -ex 'x/50gx $sp' -args depmod -b "$BUILDROOT" "$KERNELVERSION"
```
(in `/usr/lib/initcpio/functions` and similar in `/usr/share/libalpm/scripts/depmod`) I get the following crash information:

```
Program received signal SIGSEGV, Segmentation fault.
0x000055555555943e in index_insert (node=0x555555fad950, 
    node@entry=0x555555fe4d60, 
    key=key@entry=0x7fffffffa9e0 "symbol:", <incomplete sequence \375>, 
    value=<optimized out>, priority=<optimized out>) at tools/depmod.c:290
290					prefix[j] = '\0';
#0  0x000055555555943e in index_insert (node=0x555555fad950, 
    node@entry=0x555555fe4d60, 
    key=key@entry=0x7fffffffa9e0 "symbol:", <incomplete sequence \375>, 
    value=<optimized out>, priority=<optimized out>) at tools/depmod.c:290
        prefix = 0x7ffff79f0000 <argp_default_options+96> "m4\233\367\377\177"
        n = <optimized out>
        j = 0
        child = <optimized out>
        i = <optimized out>
        ch = 109
#1  0x000055555555b9a4 in output_symbols_bin (depmod=0x7fffffffaf90, 
    out=0x55555558a4f0) at tools/depmod.c:2352
        duplicate = <optimized out>
        sym = 0x5555556fefc0
        len = 2
        idx = 0x555555fe4d60
        alias = "symbol:\026\375\000\a\265\006\00055]\202\216(\234E\021I\243\235\020\006\000\177\000\000\002\000\000\000\000\000\000\000J\262\000 \000\000\000\000\020\307\aVUU\000\000\002\000\000\000\000\000\000\000\002\000\000\000\000\000\000\0001f\211\367\377\177\000\000\360\244XUUU\000\000\360\244XUUU\000\000\004\000\000\000\000\000\000\0001f\211\367\377\177\000\000\004\000\000\000\000\000\000\000x^\211\367\377\177\000\000\034\265\aVUU\000\000:\000\000\000\000\000\000\000 \344\236\367\377\177\000\000\000\000\000\000\000\000\000\000\001\000\000\000\000\000\000\0001f\211\367\377\177\000\000\360\244XUUU\000\000\360\244XUUU\000\000\004\000\000\000\000\000\000\000"...
        salias = {
          bytes = 0x7fffffffa9e0 "symbol:", <incomplete sequence \375>, 
          size = 1024, need_free = false}
        baselen = 7
        iter = {hash = 0x5555555824d0, bucket = 206, entry = 3}
        v = 0x5555556fefc0
        ret = 0
#2  0x00005555555678e2 in depmod_output (out=0x0, depmod=0x7fffffffaf90)
    at tools/depmod.c:2622
        fp = 0x55555558a4f0
        tmp = "modules.symbols.bin.35708.970794.1678954917", '\000' <repeats 211 times>
        r = <optimized out>
        ferr = <optimized out>
        dname = 0x7fffffffb208 "/tmp/mkinitcpio.Q7ebwH/root/lib/modules/6.2.6-arch1-1-dbg"
        dfd = <optimized out>
        err = 0
        tv = {tv_sec = 1678954917, tv_usec = 970794}
        itr = <optimized out>
        depfiles = <optimized out>
        dname = <optimized out>
        dfd = <optimized out>
        err = <optimized out>
        tv = <optimized out>
        fp = <optimized out>
        tmp = <optimized out>
[... 343 lines cut ...]
0x7fffffffa9a0:	0x00007fffffffa9f0	0x0000555558892c50
0x7fffffffa9b0:	0x000055555938e6ac	0x0000555556784750
0x7fffffffa9c0:	0x0000555555582770	0x0000000800000017
0x7fffffffa9d0:	0x00007fffffffa9f0	0x0000000000000400
0x7fffffffa9e0:	0x0000555558892c00	0x0000555558892c50
0x7fffffffa9f0:	0xf23a6c6f626d7973	0x374a3d3d4a80003d
0x7fffffffaa00:	0x2b5662b471586a8e	0x00007f000617a0fa
0x7fffffffaa10:	0x000055555939b6fc	0x0000000000000029
0x7fffffffaa20:	0x00007ffff79ee420	0x0000000000000004
0x7fffffffaa30:	0x0000000000000005	0x00007ffff7896631
0x7fffffffaa40:	0x0000555558892c50	0x0000555558892c50
0x7fffffffaa50:	0x0000000000000004	0x00007ffff7896631
0x7fffffffaa60:	0x0000000000000004	0x00007ffff7895e78
0x7fffffffaa70:	0x00005555593815bc	0x0000000000000026
0x7fffffffaa80:	0x00007ffff79ee420	0x0000000000000008
0x7fffffffaa90:	0x0000000000000009	0x00007ffff7896631
0x7fffffffaaa0:	0x0000555558892c50	0x0000555558892c50
```

I could upload the `linux-dbg-6.2.6.arch1-1-x86_64.pkg.tar.zst` package somewhere if necessary.

Thanks!

Best
Martin

evelikov, 2024-11-16:
While I was looking into this I noticed that all the (relevant) sections were present but were missing the correct flag(s) (IIRC `CONTENTS` as listed by objdump). Where kmod as a whole never checks those and blindly assumes there is data within the respective sections.

Since these are stripped debug-only modules, they cannot be used as-is. If they truly are invalid maybe we should report a bug against the STRIP tool used?

lucasdemarchi, 2025-02-25:
@stoeckmann is this fixed by the commits merged or still depend on that one commit that is pending?

stoeckmann, 2025-02-25:
@lucasdemarchi, the issue is fixed as good as possible. The supplied module files lack required sections by the kernel, so they cannot be loaded. But kmod does not segfault anymore, either.

So, yes. This issue can be closed.

lucasdemarchi, 2025-02-25:
@stoeckmann thanks

closed 2025-02-25 by lucasdemarchi