kmod-project/kmod #33: When using multiple softdep for a module, only one of them have effects

yshui, 2024-01-11:
Comments in the code suggest this might be intentional. But this is a surprising behavior. In addition to that, when there is already `softdep` commands in system's `modules.softdep`, user configured `softdep`s  in `/etc/modprobe.d` will just be ignored, making it effectively non-user configurable. I think this is undesirable.

lucasdemarchi, 2024-01-11:
> When using multiple softdep for a module, only one of them have effects

that's correct.  The config  parsing in kmod is designed so one config overrides a previously set one. 

>  user configured softdeps in /etc/modprobe.d will just be ignored, making it effectively non-user configurable. I think this is undesirable.

that would be a bug... we should be respecting the precedence order as per MODPROBE.D(5) :
/lib/modprobe.d/*.conf
/usr/local/lib/modprobe.d/*.conf
/run/modprobe.d/*.conf
/etc/modprobe.d/*.conf

... with the latest files overriding what is set in the previous ones.

Improvements we could add to kmod:

1) Define a different syntax to extend a previous setting. so it does what you were expecting. systemd's  units  use a `+=`  for that. We could do something similar
2) Allow a setting to be removed (i.e. it should be possible to set a softdep to "empty" after it has been set by another file).

Thoughts?

lucasdemarchi, 2025-02-24:
We should add a module to the testsuite with 2 softdep and then make sure all of them are loaded. Another tests would be to have 2 softdeps in different directories.

closed 2026-11-02 by lucasdemarchi