kmod-project/kmod #40: Add a mask command or don't deprecate the install command for modprobe.d

danielzgtg, 2024-05-02:
I am currently using and recommending syntax like `install nouveau /bin/false` . No, `blacklist nouveau` is not enough. Reading the manpages ([albeit unchanged 13 years](https://github.com/kmod-project/kmod/commit/5173e8e842e0c3ab5dfb78944a747e956e3b6670)), I am concerned about its removal plans:

> The long term future of this command as a solution to the problem of providing additional module dependencies is not assured and it is intended to replace this command with a warning about its eventual removal or deprecation at some point in a future release.

The kmod-systemd correspondences look like

| kmod | systemd |
| --- | --- |
| `blacklist modulename` | `systemctl disable example.service` |
| `install modulename /bin/false` | `systemctl mask example.service` |

The problem is similar. `blacklist modulename` doesn't guarantee the module doesn't load, just like how `systemctl disable` doesn't actually guarantee the service doesn't start. There's the problem how both `systemctl start example` and `sudo modprobe modulename` still work. `systemctl mask` is frequently recommended, and similarly I recommend `install nouveau /bin/false` plus a `update-initramfs -u`. Only like this does the start/load error out completely.

It's not only the `nouveau` driver needing this. I also had to use this for other drivers like `ipts` and `bluetooth`. Given I don't actually use it for "providing additional module dependencies," but for guaranteeing certain modules don't load, you could remove just `install`'s "dependency information" part without removing the whole thing.

Potential solutions:

- Special-case allowing making `/bin/install` the `command`
- Change `blacklist`'s behavior from `systemctl disable` to `systemctl mask`
- Add a `mask` command
- Remove the stale deprecation notice from `man modprobe.d`

danielzgtg, 2024-06-18:
> please provide a separate bug report

That is not a bug and is separate from my request. The behavior is by design and documented:

> Modules can contain their own aliases: usually these are aliases describing the devices they support, such as "pci:123...". These "internal" aliases can be overridden by normal "alias" keywords, but there are cases where two or more modules both support the same devices, or a module invalidly claims to support a device that it does not: the blacklist keyword indicates that all of that particular module's internal aliases are to be ignored.

`blacklist` seems to successfully prevent a device's plug-and-play/probe from loading the module. However, that is one one of the two places where the module could be loaded.

What I want to block is ***explicit*** module loading via `sudo modprobe nouveau`. There is something somewhere, perhaps a buggy display manager or other systemd service, that is trying to run a modprobe itself. My use case for `modprobe.d` is to throw a monkeywrench into the misbehaving service's attempts at loading nouveau.

lucasdemarchi, 2024-06-18:
I thought before about adding a "hard-blacklist" with the behavior you mentioned. I like your comparison with systemd services and the use of "mask" keyword.... It makes total sense and would be a good addition to kmod.  If you'd like to contribute such a change, feel free to send it here or in the mailing list. Otherwise I will look at adding it soon. Thanks.

howardthuang, 2025-03-03:
Kmod Team 1 would like to claim this issue for Hackathon.

closed 2026-10-06 by lucasdemarchi