kmod-project/kmod #50: testsuite: Allow to run tests under a vm

lucasdemarchi, 2024-07-17:
Instead o mocking the kernel side, allow to run the tests with a real kernel under a vm. This means removing the LD_PRELOAD where needed and integrate the VM setup to be executed under qemu.

gongjun-song, 2025-03-04:
Could you please clarify our goal? Which of the following meets your expectations?

1. Build QEMU with Meson to facilitate automatic testing of a real kernel in the testsuite.
2. Use QEMU to manually test the testsuite in the vm with a real kernel.

lucasdemarchi, 2025-03-04:
@gongjun-song it's more around (2). Currently the testsuite mimics the kernel interface by redirecting the calls to init_module/finit_module/delete_module and any path-related call to somewhere else. See `testsuite/{delete_module.c,init_module.c,path.c,uname.c}`. Goal is to be able to disable this part and test it in a real kernel.

gongjun-song, 2025-03-06:
Kmod Team 2 will work on this issue for the Hackathon, thanks!