kmod-project/kmod #220: summary: List buildtype and strip options

dileks, 2024-11-01:
Hi,

I use a custom native INI file ` build-dileks.ini `.

```
; SPDX-FileCopyrightText: 2024 Emil Velikov <emil.l.velikov@gmail.com>
; SPDX-FileCopyrightText: 2024 Lucas De Marchi <lucas.de.marchi@gmail.com>
;
; SPDX-License-Identifier: LGPL-2.1-or-later

[project options]
build-tests = false
debug-messages = false
docs = false
zstd = 'enabled'
xz = 'enabled'
zlib = 'enabled'
openssl = 'enabled'
werror = true
strip = true

[built-in options]
buildtype = 'release'
```
It might be user-friendly to list `buildtype` and `strip` in the `Options` section of the summary.

evelikov, 2024-11-01:
Have you looked at `meson configure ..../build`? It already provides all the options alongside their current and possible values.

In addition meson already prints a nice "User defined options" section when options are set on the command line. For example `meson setup --native-file build-dev.ini -D strip=false build/` will produce:

```
  User defined options
    Native files     : build-dev.ini
    strip            : false
```

Although it would be great if meson also lists the options that have been set within the native file itself.

Sounds like it will address your concern plus it will work for every project using meson, without changing anything in the individual projects.

lucasdemarchi, 2024-11-01:
@evelikov  I think what would be nice would be to have a build-release.ini setting the "recommended" options. Of course a distro could set whatever it wants, but just making sure it serves as documentation for not turning on arbitrary debug options would be great.

I think I'd add it like this. Thoughts?

```
; SPDX-FileCopyrightText: 2024 Emil Velikov <emil.l.velikov@gmail.com>
; SPDX-FileCopyrightText: 2024 Lucas De Marchi <lucas.de.marchi@gmail.com>
;
; SPDX-License-Identifier: LGPL-2.1-or-later

[project options]
build-tests = true
debug-messages = false
docs = true
zstd = 'enabled'
xz = 'enabled'
zlib = 'enabled'
openssl = 'enabled'
werror = false

[built-in options]
buildtype = 'release'
```

dileks, 2024-11-02:
@lucasdemarchi @evelikov 

Yes, a `release` INI file sounds good to me.

But I like:
```
build-tests = false
docs = false
strip = true
```

Does a `user` wants to run `build-tests` - IMHO something for a `developer`.

No strong opinion about ẁerror`setting (in my custom INI it was always `true`).

`docs` as `true` I avoided as it needs `gtk-doc-tools`  Debian package to be installed.
manpages are enough I think.

Put `developer` and `release` INI files into a seperate `ini`directory - and use `release` INI as default for building.

INFO: While my first experiments with building a selfmade kmod - kmod binary was NOT stripped.

evelikov, 2024-11-04:
> Although it would be great if meson also lists the options that have been set within the native file itself.

Opened a meson RFE, since it would be beneficial to every project https://github.com/mesonbuild/meson/issues/13865

> for not turning on arbitrary debug options would be great.

Already convinced on distro/maintainer to toggle that off \o/

evelikov, 2024-11-04:
> INFO: While my first experiments with building a selfmade kmod - kmod binary was NOT stripped.

This is the default, out of the box, experience for all projects and build systems that I know of. The general principle is that it's handled as follow-up on as needed basis. Think distributions with separate debug packages, or hitting a bug/crash (touch wood) with self-build stripped binaries.

dileks, 2024-11-04:
@evelikov 

Thanks for taking care!

closed 2024-11-07 by lucasdemarchi