kmod-project/kmod #292: Autotools dist archive for v34.

milkylainen, 2025-02-23:
Hi.

Picked up v34. Still haven't converted our build from autotools to meson.
But saw that it was slated for drop next release, so I thought I'd leave it be for now.
Anyway. Commit 651b901cf1fad4acd15b601fb9194236a7af54fb indicates the intention was dist tarball generation for this release.
But the v34 tar.xz picked up from a kernel mirror, md5 hash 3e6c5c9ad9c7367ab9c3cc4f08dfde62 doesn't seem to contain a properly generated one. Atleast it looks that way to me.
Not a big deal, but thought you should know.

lucasdemarchi, 2025-02-24:
Can you show what exactly doesn't work in the tarball? This works for me:

```
$ curl -JOL https://www.kernel.org/pub/linux/utils/kernel/kmod/kmod-34.tar.xz
$ md5sum kmod-34.tar.xz 
3e6c5c9ad9c7367ab9c3cc4f08dfde62  kmod-34.tar.xz
$ ./configure  && make -j$(nproc)
$ make DESTDIR=/tmp/inst install && tree /tmp/inst/
/tmp/inst/
└── usr
    ├── bin
    │   └── kmod
    ├── include
    │   └── libkmod.h
    ├── lib
    │   ├── libkmod.la
    │   ├── libkmod.so -> libkmod.so.2.5.1
    │   ├── libkmod.so.2 -> libkmod.so.2.5.1
    │   ├── libkmod.so.2.5.1
    │   └── pkgconfig
    │       └── libkmod.pc
    ├── sbin
    │   ├── depmod -> ../bin/kmod
    │   ├── insmod -> ../bin/kmod
    │   ├── lsmod -> ../bin/kmod
    │   ├── modinfo -> ../bin/kmod
    │   ├── modprobe -> ../bin/kmod
    │   └── rmmod -> ../bin/kmod
    └── share
        ├── bash-completion
        │   └── completions
        │       ├── insmod
        │       ├── kmod
        │       ├── lsmod
        │       └── rmmod
        ├── fish
        │   └── vendor_functions.d
        │       ├── insmod.fish
        │       ├── lsmod.fish
        │       └── rmmod.fish
        ├── man
        │   ├── man5
        │   │   ├── depmod.d.5
        │   │   ├── modprobe.d.5
        │   │   ├── modules.dep.5
        │   │   └── modules.dep.bin.5
        │   └── man8
        │       ├── depmod.8
        │       ├── insmod.8
        │       ├── kmod.8
        │       ├── lsmod.8
        │       ├── modinfo.8
        │       ├── modprobe.8
        │       └── rmmod.8
        ├── pkgconfig
        │   └── kmod.pc
        └── zsh
            └── site-functions
                ├── _insmod
                ├── _lsmod
                └── _rmmod

18 directories, 35 files
```

abelbeck, 2025-03-25:
I also see this issue cross-compiling via Buildroot.  The symlinks in 34 ``build-aux/`` appear to be part of the problem.

kmod 33 builds fine:
```
$ make kmod-patch
>>> kmod 33 Extracting
xzcat /home/dev/astlinux/trunk/dl/kmod-33.tar.xz | tar --strip-components=1 -C /home/dev/astlinux/trunk/output/build/kmod-33  -xf -
>>> kmod 33 Patching package/kmod
for file in config.guess config.sub; do for i in $(find /home/dev/astlinux/trunk/output/build/kmod-33 -name $file); do cp package/gnuconfig/$file $i; done; done
>>> kmod 33 Patching libtool

$ ls -l output/build/kmod-33/build-aux/*
-rwxr-xr-x 1 dev dev   7621 Jul 20  2024 output/build/kmod-33/build-aux/compile
-rwxr-xr-x 1 dev dev  44747 Mar 25 10:18 output/build/kmod-33/build-aux/config.guess
-rwxr-xr-x 1 dev dev  35171 Mar 25 10:18 output/build/kmod-33/build-aux/config.sub
-rwxr-xr-x 1 dev dev  23695 Jul 20  2024 output/build/kmod-33/build-aux/depcomp
-rwxr-xr-x 1 dev dev  15437 Jul 20  2024 output/build/kmod-33/build-aux/install-sh
-rw-r--r-- 1 dev dev 331635 Aug  3  2024 output/build/kmod-33/build-aux/ltmain.sh
-rwxr-xr-x 1 dev dev   7728 Jul 20  2024 output/build/kmod-33/build-aux/missing
-rwxr-xr-x 1 dev dev   5215 Jul 20  2024 output/build/kmod-33/build-aux/test-driver
```

kmod 34 build fails:
```
$ make kmod-patch
>>> kmod 34 Extracting
xzcat /home/dev/astlinux/trunk/dl/kmod-34.tar.xz | tar --strip-components=1 -C /home/dev/astlinux/trunk/output/build/kmod-34  -xf -
>>> kmod 34 Patching package/kmod
for file in config.guess config.sub; do for i in $(find /home/dev/astlinux/trunk/output/build/kmod-34 -name $file); do cp package/gnuconfig/$file $i; done; done
cp: not writing through dangling symlink '/home/dev/astlinux/trunk/output/build/kmod-34/build-aux/config.guess'
cp: not writing through dangling symlink '/home/dev/astlinux/trunk/output/build/kmod-34/build-aux/config.sub'
make: *** [package/Makefile.package.in:264: /home/dev/astlinux/trunk/output/build/kmod-34/.stamp_patched] Error 1

$ ls -l output/build/kmod-34/build-aux/
total 0
lrwxrwxrwx 1 dev dev 32 Feb 21 14:37 compile -> /usr/share/automake-1.17/compile
lrwxrwxrwx 1 dev dev 37 Feb 21 14:37 config.guess -> /usr/share/automake-1.17/config.guess
lrwxrwxrwx 1 dev dev 35 Feb 21 14:37 config.sub -> /usr/share/automake-1.17/config.sub
lrwxrwxrwx 1 dev dev 32 Feb 21 14:37 depcomp -> /usr/share/automake-1.17/depcomp
lrwxrwxrwx 1 dev dev 35 Feb 21 14:37 install-sh -> /usr/share/automake-1.17/install-sh
lrwxrwxrwx 1 dev dev 38 Feb 21 14:37 ltmain.sh -> /usr/share/libtool/build-aux/ltmain.sh
lrwxrwxrwx 1 dev dev 32 Feb 21 14:37 missing -> /usr/share/automake-1.17/missing
lrwxrwxrwx 1 dev dev 36 Feb 21 14:37 test-driver -> /usr/share/automake-1.17/test-driver
```
kmod 34.1 build also fails.

lucasdemarchi, 2025-03-25:
@abelbeck @milkylainen can you test https://github.com/kmod-project/kmod/pull/325?

lucasdemarchi, 2025-03-25:
note that this should fix the generated tarball, it's not sufficient to apply the patch on top. I extracted both kmod-33 and kmod-34.1 (from that branch). This is the diff when comparing the output of the tree command:

```diff
diff --git a/kmod33.tree b/kmod34.1.tree
index 5e3e5a4..fd7d239 100644
--- a/kmod33.tree
+++ b/kmod34.1.tree
@@ -1,5 +1,6 @@
-kmod-33
+kmod-34.1
 ├── aclocal.m4
+├── autogen.sh
 ├── build-aux
 │   ├── compile
 │   ├── config.guess
@@ -9,10 +10,15 @@ kmod-33
 │   ├── ltmain.sh
 │   ├── missing
 │   └── test-driver
+├── build-dev.ini
+├── CODING-STYLE
 ├── config.h.in
 ├── configure
 ├── configure.ac
+├── CONTRIBUTING.md
 ├── COPYING
+├── DCO.txt
+├── gtk-doc.make
 ├── libkmod
 │   ├── COPYING
 │   ├── docs
@@ -21,6 +27,7 @@ kmod-33
 │   │   ├── libkmod-sections.txt
 │   │   ├── Makefile.am
 │   │   ├── Makefile.in
+│   │   ├── meson.build
 │   │   └── version.xml.in
 │   ├── libkmod-builtin.c
 │   ├── libkmod.c
@@ -40,6 +47,7 @@ kmod-33
 │   ├── libkmod.pc.in
 │   ├── libkmod-signature.c
 │   ├── libkmod.sym
+│   ├── Makefile
 │   └── README
 ├── m4
 │   ├── attributes.m4
@@ -53,75 +61,69 @@ kmod-33
 ├── Makefile.am
 ├── Makefile.in
 ├── man
-│   ├── depmod.8
 │   ├── depmod.8.scd
-│   ├── depmod.d.5
 │   ├── depmod.d.5.scd
-│   ├── insmod.8
 │   ├── insmod.8.scd
-│   ├── kmod.8
 │   ├── kmod.8.scd
-│   ├── lsmod.8
 │   ├── lsmod.8.scd
 │   ├── Makefile.am
 │   ├── Makefile.in
-│   ├── modinfo.8
+│   ├── meson.build
 │   ├── modinfo.8.scd
-│   ├── modprobe.8
 │   ├── modprobe.8.scd
-│   ├── modprobe.d.5
+│   ├── modprobe.conf.5
[... 227 lines cut ...]
 │   ├── test-strbuf.c
 │   ├── testsuite.c
 │   ├── testsuite.h
 │   ├── test-testsuite.c
-│   ├── test-user.c
 │   ├── test-util.c
+│   ├── test-weakdep.c
 │   └── uname.c
 └── tools
     ├── COPYING
@@ -657,9 +656,12 @@ kmod-33
     ├── log.c
     ├── log.h
     ├── lsmod.c
+    ├── Makefile
     ├── modinfo.c
     ├── modprobe.c
+    ├── opt.c
+    ├── opt.h
     ├── rmmod.c
     └── static-nodes.c
 
-205 directories, 458 files
+206 directories, 459 files
```

lucasdemarchi, 2025-03-25:
In order to test locally you can do a git checkout and create the tarball. Then you use it in your distro recipe.

```
$ meson setup build
$ meson dist -C build
```

abelbeck, 2025-03-25:
I used Debian 12, cloned the v34.1 tag
```
$ git clone --depth 1 --branch "v34.1" https://github.com/kmod-project/kmod.git "v34.1"
```
applied the patch https://patch-diff.githubusercontent.com/raw/kmod-project/kmod/pull/325.patch

The resulting ``build/meson-dist/kmod-34.1.tar.xz`` still had symlinks.
```
drwxr-xr-x  0 root   root        0 Mar 25 17:20 kmod-34.1/build-aux/
lrwxrwxrwx  0 root   root        0 Mar 25 17:20 kmod-34.1/build-aux/compile -> /usr/share/automake-1.16/compile
lrwxrwxrwx  0 root   root        0 Mar 25 17:20 kmod-34.1/build-aux/depcomp -> /usr/share/automake-1.16/depcomp
lrwxrwxrwx  0 root   root        0 Mar 25 17:20 kmod-34.1/build-aux/install-sh -> /usr/share/automake-1.16/install-sh
lrwxrwxrwx  0 root   root        0 Mar 25 17:20 kmod-34.1/build-aux/missing -> /usr/share/automake-1.16/missing
lrwxrwxrwx  0 root   root        0 Mar 25 17:20 kmod-34.1/build-aux/test-driver -> /usr/share/automake-1.16/test-driver
```

lucasdemarchi, 2025-03-26:
```
$ podman run -v $PWD:/src -w /src -it debian:12
$ # from .github/actions/setup-debian/action.yml
$ export DEBIAN_FRONTEND=noninteractive
        export TZ=Etc/UTC
        apt-get update
        apt-get install --yes \
          autoconf \
          automake \
          build-essential \
          clang \
          gcc-multilib \
          git \
          gtk-doc-tools \
          liblzma-dev \
          libssl-dev \
          libtool \
          libzstd-dev \
          linux-headers-generic \
          meson \
          scdoc \
          zlib1g-dev \
          zstd
$ git fetch https://github.com/kmod-project/kmod/ refs/pull/325/head
$ git reset --hard FETCH_HEAD
$ git clean -fxd
$  meson dist -C build 
Running custom dist script '/src/scripts/autogen-for-tarball.sh'
Testing distribution package /src/build/meson-dist/kmod-34.1.tar.xz
...
$ tar -xf /src/build/meson-dist/kmod-34.1.tar.xz
$ find kmod-34.1/ -type l
kmod-34.1/testsuite/rootfs-pristine/test-loaded/sys/module/btusb/drivers/usb:btusb
$ ls -l kmod-34.1/build-aux/*
-rwxr-xr-x 1 root root   7400 Mar 26 04:56 kmod-34.1/build-aux/compile
-rwxr-xr-x 1 root root  49482 Mar 26 04:56 kmod-34.1/build-aux/config.guess
-rwxr-xr-x 1 root root  35406 Mar 26 04:56 kmod-34.1/build-aux/config.sub
-rwxr-xr-x 1 root root  23568 Mar 26 04:56 kmod-34.1/build-aux/depcomp
-rwxr-xr-x 1 root root  15358 Mar 26 04:56 kmod-34.1/build-aux/install-sh
-rwxr-xr-x 1 root root 333057 Mar 26 04:56 kmod-34.1/build-aux/ltmain.sh
-rwxr-xr-x 1 root root   6878 Mar 26 04:56 kmod-34.1/build-aux/missing
-rwxr-xr-x 1 root root   4879 Mar 26 04:56 kmod-34.1/build-aux/test-driver
```

abelbeck, 2025-03-26:
@lucasdemarchi : The difference is I need to run ``meson setup build`` before ``meson dist -C build`` is allowed to run.

While I'm not able to generate my own ``kmod-34.1`` tarball, it looks like your tarball (with ``build-aux/`` files, not symlinks) with #325 should fix the issue.  Many thanks for your time.

lucasdemarchi, 2025-03-28:
Patch applied to kmod-34.y branch and new v34.2 release done:
https://lore.kernel.org/linux-modules/nbfzghf5ol2unxajqj2fumagq25leri3dtnlu47rsxyr63g2ul@g5zgumnoxpps/T/#u

@abelbeck @milkylainen note the hint there though:  there won't be a new release with autotools, so it´d be good if whatever you are using is migrated to meson. If you are using Debian, we even have that covered in our CI going back to Debian 11.

abelbeck, 2025-03-28:
@lucasdemarchi : v34.2 builds via autotools and works as expected.  Thanks!

BTW:  It is sad that autotools support is being dropped, as our build system does not currently support meson.  Great that v34.2 builds until meson can be added.

lucasdemarchi, 2025-03-28:
@abelbeck  is it an internal build system? I don't see how support meson would be a daunting task. Even if it's not abstracted by the tooling, it's no more than 3 commands, like with autotools. I bet you could even "simulate" autotools by having a configure script and makefile you apply as downstream patch