kmod-project/kmod #298: kmod 34 linking fails on Debian Bullseye with missing dlclose reference

HiassofT, 2025-02-25:
When trying to build kmod 34 on a Debian Bullseye system the link step fails with an undefined reference to dlclose:

```
# meson setup -Dzstd=disabled -Dxz=disabled -Dzlib=disabled -Dopenssl=enabled -Dtools=true -Dbuild-tests=false -Ddocs=false -Dmanpages=false /tmp/kmod-34
# meson compile
...
[27/33] Linking target kmod
FAILED: kmod 
cc  -o kmod kmod.p/tools_depmod.c.o kmod.p/tools_insmod.c.o kmod.p/tools_kmod.c.o kmod.p/tools_log.c.o kmod.p/tools_lsmod.c.o kmod.p/tools_modinfo.c.o kmod.p/tools_modprobe.c.o kmod.p/tools_opt.c.o kmod.p/tools_rmmod.c.o kmod.p/tools_static-nodes.c.o -Wl,--as-needed -Wl,--no-undefined -pie -Wl,--gc-sections -Wl,--start-group libshared.a libkmod-internal.a /usr/lib/x86_64-linux-gnu/libcrypto.so -Wl,--end-group
/usr/bin/ld: libshared.a.p/shared_util.c.o: undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
```

The linker line is missing `-ldl` which is needed with the older GNU toolchain.

Building on Debian Bookworm works fine and running `LDFLAGS="-ldl" meson setup ....` on Bullseye works around the issue but it should better be fixed in the build system.

Here's a simple containerfile to reproduce the issue:
```
FROM docker.io/debian:bullseye-slim AS kmod-build
RUN \
    echo "deb http://deb.debian.org/debian bullseye-backports main" >> /etc/apt/sources.list \
    && apt-get update \
    && apt-get -y install build-essential libssl-dev pkg-config curl \
    && apt-get -y -t bullseye-backports install meson ninja-build

FROM kmod-build
RUN \
    curl https://www.kernel.org/pub/linux/utils/kernel/kmod/kmod-34.tar.xz | tar Jxf - -C /tmp/ \
    && mkdir /tmp/kmod-build \
    && cd /tmp/kmod-build \
    && meson setup -Dzstd=disabled -Dxz=disabled -Dzlib=disabled -Dopenssl=enabled -Dtools=true -Dbuild-tests=false -Ddocs=false -Dmanpages=false /tmp/kmod-34 \
    && meson compile
```

HiassofT, 2025-02-25:
@lucasdemarchi thanks a lot for the quick fix!

Testing with #299 looks fine here, too, I could successfully build LibreELEC master on Debian Bullseye (PR added as a local patch)

lucasdemarchi, 2025-02-25:
@HiassofT any particular reason for those flags? Is it what's being used in LibreELEC? I'm considering adding a CI entry for covering it and wandering if I should bring all those or just keep the default.

HiassofT, 2025-02-25:
There's no particular reason for the flags, I simply tested with what we are currently using in LibreELEC https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/sysutils/kmod/package.mk#L15-L32

Adding a CI test is a good idea, using the default options should be fine I guess

lucasdemarchi, 2025-02-26:
I added it to CI, but from my local tests our testsuite is not passing there. I'm not sure yet if it's a test issue (or real problems due to different libc and/or kernel headers and/or libraries).

closed 2025-02-26 by lucasdemarchi