kmod-project/kmod #301: consider shipping pre generated man pages in release tarballs

dantob, 2025-02-28:
Having a look at some other tools, they ship pre generated man pages in release tarballs. This could help distros like linuxfromscratch that don't have scdoc available at build time.

examples:
coreutils
shadow
util-linux

lucasdemarchi, 2025-03-05:
@dantob yeah... we used to ship pre-generated manpages. But I'm not fond on doing that anymore. I'd rather not ship any generated content at all, which is better from security pov. Any example of projects using meson that ship pre-generated manpages?  Btw, one of the reasons to use scdoc for manpages is that it is so simple - I'm sure linuxfromscratch could build the extra .c files if they want the manpages.  Maybe we could even consider vendoring the tool?  @evelikov  thoughts?

dantob, 2025-03-06:
> Btw, one of the reasons to use scdoc for manpages is that it is so simple - I'm sure linuxfromscratch could build the extra .c files if they want the manpages.

scdoc is definitely buildable on stock lfs without too much extra effort.

closed 2025-03-25 by lucasdemarchi