kmod-project/kmod #338: Bus error when running kmod on architecture that does not support unaligned access

pushatik, 2025-04-15:
I encountered problem with kmod when working with bpif3 board. I used gcc 14.2.0, flags: -O2 (or -O3) -march=rv64gv.
Crash looks like this:
```
modprobe[266]: unhandled signal 7 code 0x1 at 0x0000002aba6dc9d6 in kmod[2aba6cd000+22000]

Backtrace:
#0  read_long_mm (p=<synthetic pointer>) at libkmod/libkmod-index.c:632
#1  index_mm_read_node (idx=0x2ad4e953c0, offset=536871449) at libkmod/libkmod-index.c:679
...

Instraction:
v = get_unaligned((uint32_t *) addr);
f9d6           vle32.v v2, (a2)     <-- crash here, load in 32-bit vector register 

Register:
a2             0x3f893a321b     272885232155   <-- unaligned address
```

Maybe this error is related to gcc and vector optimization, but why does kmod use get_unaligned  in this form?
```
#define get_unaligned(ptr)                       \
	({                                       \
		struct __attribute__((packed)) { \
			typeof(*(ptr)) __v;      \
		} *__p = (typeof(__p))(ptr);     \
		__p->__v;                        \
	})
```

There is no memory allocation at aligned address, and "packed" is not part of the C standard, so there is no clear description that it should help in this situation.

I wrote my own implementation of get_unaligned and put_unaligned where there is memory allocation at aligned address: 
```
#define get_unaligned(src) \
    *(typeof(*(src))*)memcpy( \
        (union {char alias[1]; typeof(*(src)) value;}){.value = 0}.alias, \
        (src), sizeof(*(src)))

#define put_unaligned(val, dst) \
    ((void)sizeof(0 ? (typeof(val)*)0 : (dst)), \    /* <-- check for same type val and (*dst)*/
     (void)memcpy((dst), \
        (union {char alias[1]; __typeof(*(dst)) value;}){.value = (val)}.alias, \
        sizeof(val)))
```
With the same compiler flags program stopped crashing.
Here are some examples where program crashes with your macros  (error is visible even in assembler code), but doesn't crash with mine:
Example 1: https://godbolt.org/z/5d1qEMbz5
Example 2: https://godbolt.org/z/cv349fY1K

I would like to know your opinion, is it possible to change implementation of get_unaligned and put_unaligned?

lucasdemarchi, 2025-04-15:
Interesting, thanks for the report.

> #0  read_long_mm (p=<synthetic pointer>) at libkmod/libkmod-index.c:632

what kmod version is that? We don't have read_long_mm() anymore.

> There is no memory allocation at aligned address, and "packed" is not part of the C standard, so there is no clear description that it should help in this situation.

The packed attribute there is simply avoiding the struct to have a size greater than the type its wrapping. It's not what is forcing the alignment.

What should be forcing the alignment here is actually the struct on the stack. The compiler should create it aligned. We then memcpy the data from addr to this aligned variable and return that.

That get_unaligned() implementation is kind of "well known to work" and used in other projects. I wonder which of the cases you are hitting:

1) the compiler is not aligning the  wrapper struct
2) the compiler is generating wrong load instructions

I wonder if this works for you?

```diff
diff --git a/shared/util.h b/shared/util.h
index c36d39e..e46d4a8 100644
--- a/shared/util.h
+++ b/shared/util.h
@@ -73,7 +73,7 @@ unsigned long long get_backoff_delta_msec(unsigned long long t0, unsigned long l
 /* ************************************************************************ */
 #define get_unaligned(ptr)                       \
        ({                                       \
-               struct __attribute__((packed)) { \
+               struct __attribute__((packed)) _aligned_((sizeof(*ptr))) { \
                        typeof(*(ptr)) __v;      \
                } *__p = (typeof(__p))(ptr);     \
                __p->__v;
```

lucasdemarchi, 2025-04-15:
Some places in which you may find issues if the compiler is not fixed:

bluez - https://github.com/IanHarvey/bluepy/blob/master/bluez-5.47/src/shared/util.h
Linux kernel, include/asm-generic/unaligned.h , but recently there were some commits moving it:
https://github.com/torvalds/linux/blob/1a1d569a75f3ab2923cb62daf356d102e4df2b86/include/linux/unaligned.h#L13

pushatik, 2025-04-16:
Thanks for quick response.

>what kmod version is that? We don't have read_long_mm() anymore.

Kmod have v33 version, but get_unaligned was not changed in master, so I have not tested kmod on newer version

>I wonder if this works for you?

Assemble code has not changed:
```
vsetivli        zero,2,e32,mf2,ta,ma
addi    a4,a5,45
vle32.v v1,0(a4)
```
Link to godbolt: https://godbolt.org/z/78Mvef1Wc

>What should be forcing the alignment here is actually the struct on the stack. The compiler should create it aligned. We then memcpy the data from addr to this aligned variable and return that.

But there is no creation of struct on stack. There is creation of pointer to struct.
Moreover, I noticed that if I use gcc-13.3 or gcc-trunk I don't get kmod crash (see assembler code):
(gcc-13.3) https://godbolt.org/z/nc1zxzaco
(gcc-trunk) https://godbolt.org/z/6f4cocrcT

Probably something changed in gcc implementation, so assembler code became different. But this is not compiler problem. There is nothing in C standard that says that code get_unaligned() will definitely work.
Maybe that something will change again in (for example) gcc-16, and existing get_unaligned() implementation will break again for certain architectures.

pushatik, 2025-04-16:
<Some places in which you may find issues if the compiler is not fixed

I can test other projects for this problem, but it seems pointless because I wrote small programs that show macro problem.

lucasdemarchi, 2025-04-17:
> Probably something changed in gcc implementation, so assembler code became different. But this is not compiler problem.

note that when you remove the loop and do a single read, the generated code is the same again. It seems there's definitely something going wrong with the loop unrolling in that gcc 14. As I said, this is a widespread implementation of reading an unaligned value. Checking on the Linux kernel side, that implementation goes back to the first commit in git history (2.6.12). And the implementation is very similar:

```
struct __una_u64 { __u64 x __attribute__((packed)); };

/*
 * Elemental unaligned loads 
 */

static inline unsigned long __uldq(const __u64 *addr)
{
        const struct __una_u64 *ptr = (const struct __una_u64 *) addr;
        return ptr->x;
}
```