kmod-project/kmod #392: format of `.bin` files is undocumented

ddiss, 2025-09-30:
Thanks for your work on this very helpful library.
Has anyone made an effort to document the format of the binary output files `modules.dep.bin`, `modules.builtin.bin`, etc? The `modules.dep` manpage makes it clear that the output format is not stable:
```
       These files are not intended for editing or use by any additional utilities as their format is subject to change in the future.
```
Nevertheless, I think it'd be helpful kmod developers and others.

ddiss, 2025-10-20:
Thanks for the response...

> The format is documented within the kmod codebase - we even implement it twice in libkmod and depmod (former of which should be enhanced and used in the latter).
> 
> That aside, I don't think that documenting it within the man page is a good idea. It's an internal implementation detail, which should remain opaque to third parties.

I agree that a man page would be the wrong spot, but perhaps some documentation elsewhere. I don't think anyone benefits from deliberate opacity, given that the on-disk-format appears stable and needs ongoing support for backwards compatibility. Besides, not having a `.bin` reference leads people to hack on inefficient parsers for the (self-explanatory) text format
E.g. https://raw.githubusercontent.com/ddiss/rapido/refs/heads/parse_depmod_awk/tools/parse_depmod.awk

evelikov, 2025-10-26:
@ddiss sounds like the real question is can we extend the kmod tools to output subset of information, so people don't screen-scape/awk/sed manually...

For what it's worth, I am fully in favour of that and I would be happy to review PRs (or write one) implementing that.

My awk is a bit rusty - what are the requirements for the linked script/project?